MangoAutomation/ma-core-public #633: Core - Copy Data Source startup issues

Puckfist, 2016-02-05:
When the "Copy Data Source" button on the Data Sources page is used, the copied data source is polling but without any points in its context. I demonstrated with one virtual source polling at ten second with one activated point, then dropping a breakpoint in the poll method and creating a copy. I get two polls in a ten second interval, but only one source has a nonempty dataPoints

terrypacker, 2016-02-10:
This brings up a few issues with how we are saving data sources.  There should be no way to save a data source or data point without calling save in the RuntimeManager.

We need to test any references to:

DataSourceDao.copyDataSource 
DataSourceDao.copyDataSourcePoints
DataSourceDao.copy

and 

DataPointDao.copy

terrypacker, 2016-02-11:
I've found that copying a data point disables it.  Is this the desired operation?

Puckfist, 2016-02-11:
Confirmed the data points were running.

closed 2016-02-11 by Puckfist