MangoAutomation/ma-core-public #634: HTTP Image Retriever image logging broken

snowwindwaverider, 2016-02-07:
An HTTP Image retriever datasource with the datapoint set to log all data does not show any images and this error shows up in the logging console. this is on a mangoES.  the MangoNoSqlPointValueDao and httpImage datasource I don't see in the repos, otherwise I would offer a patch.

`ERROR 2016-02-04 20:44:58,856 (com.serotonin.m2m2.rt.maint.BackgroundProcessing$1.run:53) - Error in work item com.serotonin.ShouldNeverHappenException: Images are not supported at com.infiniteautomation.nosql.PointValueTimeSerializer.toByteArray(PointValueTimeSerializer.java:99) at com.infiniteautomation.nosql.PointValueTimeSerializer.toByteArray(PointValueTimeSerializer.java:25) at org.tinytsdb.impl.Series.write(Series.java:139) at org.tinytsdb.impl.TinyTSDBImpl.write(TinyTSDBImpl.java:258) at com.infiniteautomation.nosql.MangoNoSqlPointValueDao.savePointValueSync(MangoNoSqlPointValueDao.java:66) at com.serotonin.m2m2.rt.dataImage.PointValueCache.savePointValue(PointValueCache.java:92) at com.serotonin.m2m2.rt.dataImage.DataPointRT.savePointValue(DataPointRT.java:303) at com.serotonin.m2m2.rt.dataImage.DataPointRT.updatePointValue(DataPointRT.java:189) at com.serotonin.m2m2.http.rt.HttpImageDataSourceRT$ImageRetriever.executeImpl(HttpImageDataSourceRT.java:191) at com.serotonin.m2m2.http.rt.HttpImageDataSourceRT$ImageRetriever.execute(HttpImageDataSourceRT.java:145) at com.serotonin.m2m2.rt.maint.BackgroundProcessing$1.run(BackgroundProcessing.java:49) at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142) at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617) at java.lang.Thread.run(Thread.java:745)`

terrypacker, 2016-02-10:
This was not implemented in the NoSQL data store, but it could be at some future date.  Using an SQL database for the point values should get you past this error.

terrypacker, 2016-04-11:
Image data type logging has been implemented.  Needing to test:
1.  saving values
2. reading values on legacy data point details page
3. reading values on new data point details page
4. Purging the web/filedata folder

Puckfist, 2016-04-15:
Getting stack trace on the new data point details:
[imageNewDPD_stack_trace.txt](https://github.com/infiniteautomation/ma-core-public/files/221827/imageNewDPD_stack_trace.txt)

Puckfist, 2016-04-15:
Hmm. Other issues. I can't seem to locate where those files are getting saved to, for sake of checking purging and whatnot. As I look into the code, I see FILEDATA_PATH only in the SystemSettingsDao and Common, and in no case does it seem configurable. The default path is ~/WEB-INF/filedata

..Did this ever work on Windows?

Puckfist, 2016-04-15:
Ah, the ~ is replaced, and I guess it is intentionally difficult to set somewhere other than default. It does log for me, in the web/WEB-INF/filedata as intended. I didn't notice the default was 'Do Not Log'!

Puckfist, 2016-04-15:
Purging does not seem to delete the image files. The point displays as though it has no value, but the web/WEB-INF/filedata folder still has the files in it.

Puckfist, 2016-04-18:
Confirmed it does not purge on the data source's purge all or that data point's purge all.

terrypacker, 2016-04-18:
That is the way file data has always been purged.  It gets removed on nightly purges only when the DataPurge task is run.  If there is no corresponding point value in the database for that image, the file gets deleted.

closed 2016-05-10 by Puckfist