MangoAutomation/ma-core-public #637: Brewers Dashboard multiple recipes with the same name

Joelhaggar, 2016-02-09:
It's possible to create multiple recipes with the same name in the database which cases a problem in the rest of the system.

terrypacker, 2016-02-10:
What version are you using?  I've looked at the code and as far as I can tell the existing recipe should be updated if you try and create one with the same name.

terrypacker, 2016-02-10:
That code has been there for a long time, so I guess it would be better to see how you are doing that.  Can you provide the steps to reproduce this?

Puckfist, 2016-02-11:
Isn't there a very unlikely race? Two users could submit two new recipes with the same name at the same moment and it would fail as written. Could simply synchronize this block?

Joelhaggar, 2016-02-11:
We don't know how this happened but we suspect it was from the user double clicking the button.

Puckfist, 2016-02-26:
This shouldn't be possible any more.

closed 2016-03-14 by Puckfist