MangoAutomation/ma-core-public #638: Rejected Execution Events for All Timer Tasks

terrypacker, 2016-02-09:
In an effort to keep the 2.7.4 release compatible with 2.7.x we have left off implementing the taskRejected method for TimerTasks.  All classes in Mango, including modules that extend TimerTask should be modified to either extend TimeoutTask or implement the rejected() method from TimerTask and raise the appropriate event.

terrypacker, 2016-02-09:
Also note that both general Runnable and ScheduledRunnable types can be rejected by the OrderedRealTimeTimer but they are currently not handled in the 'rejection logic'.  These changes will require a new core too.  For now the rejection is just logged.

terrypacker, 2016-02-10:
Note also that WorkItems should be adjusted to handle the rejected execution.

terrypacker, 2016-03-04:
This is very important for the NoSQL BatchWriteBehindManager class.

Puckfist, 2016-06-03:
Is the comment from Feb 9 still relevant? Just beginning this, I noticed there are messages from rejected interval logging, and the method for DataPointRT:EventNotifyWorkItem:rejected is a no-op

terrypacker, 2016-06-03:
I have implemented all the rejected executions for work items, so the comment from Feb 9 is finished.  If you take a look at OrderedWorkItemRunnable.rejected() you can see why the DataPointRT:EventNotifyWorkItem:rejected is a no-op.  Basically the method for the work items is to do anything special for that work item, the rejectionHandler processes all rejections via the wrapper class's rejected method.

Puckfist, 2016-11-11:
I believe this has happened? Or, it is at least using the reflection until 2.9?

terrypacker, 2016-11-11:
Yes leave this open as we will need to revisit when we get into the code in the 2.9.x branch as there are several features we may want to bring in.

closed 2017-03-20 by Puckfist