MangoAutomation/ma-core-public #640: Script Permissions - Non-admin Editing & Permissions

Puckfist, 2016-02-11:
If a data source (but probably point links and others as well) has script permissions fields which contain permissions the current user does not have, a non-admin cannot save that datasource even if they have edit permission for the data source, since they do not belong to the script permissions. They can delete these permissions, then save the data source, though.

terrypacker, 2016-02-11:
This will be fixed by brining in the new code from the GraphicalView that validates added and removed permissions into the Permissions class (as well as the new i18 message).  This will require a code change in all modules to use this feature as the method signature will change for the method:
```
public static void validateAddedPermissions(String itemPermissions, User user, ProcessResult response, String contextKey)
```
to:
```
public static void validateAddedPermissions(String existingPermissions, String newPermissions, User user, ProcessResult response, String contextKey)
```

terrypacker, 2016-05-26:
I am seeing a need for the permissions fields to be moved into a lower level abstract VO class.  We can also place this validation code in there too.  The publishers do not have any permissions currently and this type of change would enforce permissions on them and all the other VO classes.  For that reason I am pushing this off till later.

Puckfist, 2017-04-19:
There is an issue currently where a nonadmin with data source edit permissions can create scripting data sources with superadmin privileges.

Puckfist, 2017-04-19:
It seems script permissions validation should ensure 1) No permissions group the user doesn't have permissions to is added or removed

Anything else?