MangoAutomation/ma-core-public #641: Core - Non-admin Create Data Source, oops its gone!

Puckfist, 2016-02-11:
A user with access to the Data Source page (but not an admin) has permissions to create a Data Source. By default, there are no edit data source permissions. If they click away from that edit pane, they cannot get it back. But, interestingly, they could save it a second time and give themselves permission if they haven't lost the edit pane yet.

terrypacker, 2016-05-26:
We could add some code in the DWR that checks to see if the user is an admin and if the data source has their permission.  If not it could add a permission that they do have...

Puckfist, 2017-04-19:
Required that the user saving a data source had edit permission to that data source after the save.

closed 2017-04-19 by Puckfist