MangoAutomation/ma-core-public #644: Modbus Publisher doesn't work well with multiple publishers - needs testing

Joelhaggar, 2016-02-11:


terrypacker, 2016-02-23:
It looks like the logic for registering a new slave is incorrect as it won't start the slave if it is new.  Also we will need to review the logic for removing a ProcessImage when the publisher is unRegistered.  It looks like there currently no way to remove the ProcessImage once it has been added to the slave.

terrypacker, 2016-02-23:
Further testing should be done to ensure that the process images within the ModbusSlaveSet don't cause concurrency problems.  There is no synchronization of the map of ProcessImages and from what I can tell they can be accessed by different threads. This was lightly tested and I saw no ill effects, but...

Puckfist, 2016-02-24:
Don't we know it CAN fail, then, since that code is called in UI threads? I verified that:

1) multiple publishers can run on the same port with different slave IDs.
and...
2) publishers cannot be queried after they're stopped (without the process image removed, this was possible)

Puckfist, 2017-08-07:
A place to find failure is this loop in ModbusSlaveManager:

for(ModbusIpSlave slave : instance.ipSlaves){
				if(slave.isCompatible(rt)){
					slave.register(rt,processImage);
					registered = true;
					break;
				}else{
					canCreate = false;
					break;
				}
			}

so the failure behaviors will mostly fall upon second and after activated publishers. For instance, you can enable a second publisher with a slave ID and port collision, then the second enabled will be the one used. Or, you could then disable the second publisher and it will remove the process image and the first will not be queryable

Bug still exists, confirmed.

Puckfist, 2017-08-14:
I'm not really sure what that code is trying to achieve. I assume that the logic its looking for is something like, don't let a modbus publisher start if there is either another publisher on the port that a) has a different encapsulation value or b) has the same slave id. This registering with the compatible slave seems weird. Compatibility seems like it should be on the same port and then not have those conditions, rather than also being compatible simply should the ports not match. Also it must be non-conflicting with all existing publishers...

Puckfist, 2017-08-18:
Okay I think I figured it out, compatible is supposed to mean same port, same settings, no existing slave. Incompatible means same port different settings or existing slave

Slaves are sharing process images because slaves hold ports and process images hold slave ids. Definitely were bugs around this.

closed 2018-04-06 by terrypacker