MangoAutomation/ma-core-public #645: Core - Deleting Users, Corrupting Email Handlers

Puckfist, 2016-02-12:
If you create a user 'deletable' and add them to an email event handler, then delete the user, the handler is corrupted. It will 1) server error at eventHandler export, 2) server error at saving this event handler, 3) server error at handler delete (but it does delete and gets export working again).

Puckfist, 2016-02-16:
The offending code for export is in RecipientListEntryBean, and the problem also applies to deleting mailing lists. Null check, 77 and 79.

Solving the whole issue seems trickier as it's blobbed data. We could just write an integrity check that can be run through the interface (perhaps), rather than try to check for user or mailing list deletions all the time.

terrypacker, 2016-02-17:
I found that by adding the null checks you still end up with an empty JSON Object for the recipient in the export, which is invalid.  Instead I've added a method to clean the recipients list into the RecipientListEntryBean class.  This is now used in the following classes when deserializing the bean from the database.  This method is used in ExcelReportTemplateVO,ReportVO,EventHandlerVO.

Puckfist, 2016-02-18:
Failed on deleted mailing list. The mailing list had a user on it that was deleted before the mailing list was deleted (but I think that is irrelevant). Stack Trace:

WARN  2016-02-18 10:54:43,976 (org.directwebremoting.util.CommonsLoggingOutput.warn:67) - Method execution failed:  
com.serotonin.ShouldNeverHappenException: org.springframework.dao.EmptyResultDataAccessException: Incorrect result size: expected 1, actual 0
	at com.serotonin.util.SerializationHelper.readObjectInContext(SerializationHelper.java:93)
	at com.serotonin.m2m2.db.dao.EventDao$EventHandlerRowMapper.mapRow(EventDao.java:636)
	at com.serotonin.m2m2.db.dao.EventDao$EventHandlerRowMapper.mapRow(EventDao.java:633)
	at org.springframework.jdbc.core.RowMapperResultSetExtractor.extractData(RowMapperResultSetExtractor.java:93)
	at org.springframework.jdbc.core.RowMapperResultSetExtractor.extractData(RowMapperResultSetExtractor.java:60)
	at org.springframework.jdbc.core.JdbcTemplate$1.doInPreparedStatement(JdbcTemplate.java:693)
	at org.springframework.jdbc.core.JdbcTemplate.execute(JdbcTemplate.java:629)
	at org.springframework.jdbc.core.JdbcTemplate.query(JdbcTemplate.java:680)
	at org.springframework.jdbc.core.JdbcTemplate.query(JdbcTemplate.java:712)
	at org.springframework.jdbc.core.JdbcTemplate.query(JdbcTemplate.java:722)
	at org.springframework.jdbc.core.JdbcTemplate.query(JdbcTemplate.java:772)
	at com.serotonin.db.DaoUtils.query(DaoUtils.java:308)
	at com.serotonin.m2m2.db.dao.EventDao.getEventHandlers(EventDao.java:617)
	at com.serotonin.m2m2.db.dao.EventDao.getEventHandlers(EventDao.java:594)
	at com.serotonin.m2m2.web.dwr.EventHandlersDwr.getInitData(EventHandlersDwr.java:87)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.directwebremoting.impl.ExecuteAjaxFilter.doFilter(ExecuteAjaxFilter.java:34)
	at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428)
	at com.serotonin.m2m2.web.dwr.util.TranslationsFilter.doFilter(TranslationsFilter.java:37)
	at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428)
	at com.serotonin.m2m2.web.dwr.util.ExceptionDetectionFilter.doFilter(ExceptionDetectionFilter.java:26)
	at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428)
	at com.serotonin.m2m2.web.dwr.util.DwrPermissionFilter.doFilter(DwrPermissionFilter.java:45)
	at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428)
	at org.directwebremoting.impl.DefaultRemoter.execute(DefaultRemoter.java:431)
	at org.directwebremoting.impl.DefaultRemoter.execute(DefaultRemoter.java:283)
	at org.directwebremoting.servlet.PlainCallHandler.handle(PlainCallHandler.java:52)
	at org.directwebremoting.servlet.UrlProcessor.handle(UrlProcessor.java:101)
	at org.directwebremoting.servlet.DwrServlet.doPost(DwrServlet.java:146)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:707)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:808)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1669)
	at org.eclipse.jetty.servlets.UserAgentFilter.doFilter(UserAgentFilter.java:83)
	at org.eclipse.jetty.servlets.GzipFilter.doFilter(GzipFilter.java:300)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1652)
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:585)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:143)
	at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:577)
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:223)
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1127)
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:515)
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185)
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1061)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141)
	at org.eclipse.jetty.server.handler.ContextHandlerCollection.handle(ContextHandlerCollection.java:215)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:97)
	at org.eclipse.jetty.server.Server.handle(Server.java:499)
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:310)
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:257)
	at org.eclipse.jetty.io.AbstractConnection$2.run(AbstractConnection.java:540)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:635)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:555)
	at java.lang.Thread.run(Unknown Source)
Caused by: org.springframework.dao.EmptyResultDataAccessException: Incorrect result size: expected 1, actual 0
	at org.springframework.dao.support.DataAccessUtils.requiredSingleResult(DataAccessUtils.java:71)
	at org.springframework.jdbc.core.JdbcTemplate.queryForObject(JdbcTemplate.java:791)
	at com.serotonin.db.DaoUtils.queryForObject(DaoUtils.java:328)
	at com.serotonin.m2m2.db.dao.MailingListDao.getMailingList(MailingListDao.java:51)
[... cut ...]
	at java.io.ObjectInputStream.readSerialData(Unknown Source)
	at java.io.ObjectInputStream.readOrdinaryObject(Unknown Source)
	at java.io.ObjectInputStream.readObject0(Unknown Source)
	at java.io.ObjectInputStream.readObject(Unknown Source)
	at com.serotonin.util.SerializationHelper.readObjectInContext(SerializationHelper.java:90)
	... 56 more

It corrupted my ReportVO, too, and I cannot export it. I'm guessing the cleanRecipientList needs to use a different method to check if the mailing list exists or catch the exception as the sign.

closed 2016-02-19 by Puckfist