MangoAutomation/ma-core-public #653: DataProvider.js Bug in Safari Web Browser

terrypacker, 2016-02-19:
In DataProvider.js (2.0) in the onLoad method there is this code:

I’m seeing that Safari and Chrome show different values for the ‘arguments’ object used in this method and I think that since we are iterating over that via ‘in’ we are picking up some odd behaviour.  In chrome it seems to show as the object we are expecting but in Safari its an array of length 2.  It iterates over this array using 0,1,length for resolved on each corresponding pass. I’ve checked and it does this even without the development tools open in Safari.

Here is a screenshot of on Chrome:
[image]

Here is one on Safari
[image]

stevededrick, 2016-02-22:
see: Point details page amCharts doesn't display on iOS devices #565

Puckfist, 2016-02-26:
In so far as this was the issue with the charts on the HTML5 pages, confirmed fixed. The title of the issue is very nonspecific.

closed 2016-03-04 by terrypacker