MangoAutomation/ma-core-public #654: Core - Import database schema number?

Puckfist, 2016-02-22:
It is possible to update the database schema number through the import/export page. This is true for all the modules' schema versions as well. I imagine we'd rather the database scripts were handling this? Mango can fail to boot if the upgrade scripts fail, and they can fail if the attempt to run on a database that has already been upgraded. This could still be modified from the SQL console, but it could save someone from importing and old system to a new Mango and failing subsequent database upgrades.

terrypacker, 2016-03-04:
We need a blacklist for System Settings as they are just imported/export as a Map of the table.  The blacklist would ensure those settings don't get exported/imported.

terrypacker, 2016-05-26:
Fixed by excluding the import/export of any system setting that starts with databaseSchemaVersion.  That is the prefix they all use.

closed 2016-05-27 by Puckfist