MangoAutomation/ma-core-public #661: Event reset threshold

AldoRamos, 2016-02-29:
Mango allows a minimum amount of time before an event is generated; is there a way to implement a hysteresis-mitigation algorithm, so it won't consider a new event unless the condition has NOT been met for t time? Should I create a meta point to determine such fluctuation on borderline inputs?

For example, we have a wet floor sensor that as the water dries will vacillate between reporting wet/dry, generating potentially thousands of "new" events as the result of transitioning back & forth.

terrypacker, 2016-06-02:
Implement as a delay on both states, currently we only have a delay on the detection state.

Puckfist, 2018-09-12:
In my exploring implementation, I decided the event detector should still be detecting the event state the whole time, and then if the quiescent period elapses and the detector would have been active because its condition is met (read: the duration to raise the event doesn't begin at the end of quiescence) the quiescent task will raise the event.

Puckfist, 2018-09-12:
https://github.com/infiniteautomation/ma-core-public/commit/0a890b2323c84a67e1fbe1fa09e57715e455afc8

Puckfist, 2018-09-12:
Ready for testing in the quiescentEvents branch. Probably missing from API models and whatnot, though

Puckfist, 2018-09-12:
also it was my judgement the quiescent period shouldn't begin for events that return to normal until they have returned to normal, such that if they were active for longer than the quiescent period they would still have a quiescent period (and they cannot be raised again while still active, anyway)