MangoAutomation/ma-core-public #667: Point Hierarchy Rest Endpoint is slow

terrypacker, 2016-03-04:
It was found that on HTML5 Data Point Details page a the model was making SQL requests for data sources and was very slow on systems with a large number of data points.

terrypacker, 2016-03-04:
Ready for testing, the internal model code has changed and we need to ensure nothing else will have an issue with this.  The model structure is the same.

Puckfist, 2016-06-08:
I'm sure everything is relative, but it took me about 3 minutes to load a point hierarchy with 100,000 points, and simultaneously hitting that hierarchy once from swagger, once from the point hierarchy, and once from the watchlist gave a result first in swagger(5-10min), next in watchlists (10-15min), and finally on the Point hierarchy (15-20min).

However, this code has been released for well over a month, so my thinking is that others may have commented if there was an overt error.

It took over ten minutes to load on the new Data Point Details page, at which point I killed it. The XHR error says the events websocket closed while loading.

terrypacker, 2016-10-28:
Closing as ok for now.

closed 2016-10-28 by terrypacker