MangoAutomation/ma-core-public #670: Reports - Copy report duplicates XID

terrypacker, 2016-03-08:
XIDs can apparently be duplicated in the reports table by copying a report.

terrypacker, 2016-04-13:
Fixed by generating new XID during copy process.

Not changing the schema to enforce XID uniqueness as this may cause problems for conversions.  Thoughts?

Puckfist, 2016-04-15:
Hmm. Confirmed it gets a different XID. You can still manually set the XID to a duplicate. It'll export fine, but the import will rather opaquely fail with,

Incorrect result size: expected 1, actual 2
Incorrect result size: expected 1, actual 2

I wonder if it'd be possible to report the XID in this error, so that people know which need correction? Especially if we're going to permit it for legacy purposes! Something like...

ReportEmportDefinition:61
Report report = reportDao.getReport(xid);
====>
ReportVO report = null;
        try {
        	report = reportDao.getReport(xid);
        } catch (IncorrectResultSizeDataAccessException e) {
        	importContext.getResult().addGenericMessage("emport.report.sizeException", e.getMessage(), xid);
        	return;
        }

i18n.properties += emport.report.sizeException=Duplicate Xid? {0} from xid: {1}

terrypacker, 2016-05-24:
Added the above to handle situation where there are already multiple reports with same XID in the system and the user is trying to import another.

terrypacker, 2016-05-24:
- Fixing bug where a user could save a report to have the same XID as another.
- Fixing bug where report could be saved to email without recipients.

Puckfist, 2016-11-03:
Looks good!

closed 2016-11-03 by Puckfist