MangoAutomation/ma-core-public #671: Reports - Fail to load on page or export due to points

terrypacker, 2016-03-08:
Not entirely sure how this happened but a recently created report failed to load on the page, the export had a points list with no Xids...

[image]
     
    "points":[
            {
               "colour":"",
               "consolidatedChart":true,
               "plotType":1,
               "pointKey":"p21961",
               "weight":1.0
            },
            {
               "colour":"",
               "consolidatedChart":true,
               "plotType":1,
               "pointKey":"p21962",
               "weight":1.0
            },
            {
               "colour":"",
               "consolidatedChart":true,
               "plotType":1,
               "pointKey":"p21963",
               "weight":1.0
            },
            {
               "colour":"",
               "consolidatedChart":true,
               "plotType":1,
               "pointKey":"p21965",
               "weight":1.0
            }

terrypacker, 2016-05-26:
This could happen if the points for the report were deleted...

Puckfist, 2018-03-20:
https://github.com/infiniteautomation/ma-modules-public/commit/1c7c8736d3871e8716ac419045d4cddc1289bc34

closed 2018-04-05 by Puckfist