MangoAutomation/ma-core-public #674: Move MangoApi properties into env.properties

terrypacker, 2016-03-10:
This will allow accessing these properties from the core which will be useful for the JsonData web socket configuration when allowed origins for CORS are required.

Puckfist, 2016-06-07:
I will need some help defining the scope of this. I attempted to modify the env.properties such that...

rest.cors.enabled=true
rest.cors.allowedHeaders=LOGOUT

... meaning I removed 'PASSWORD' but I am still able to use the /swagger/#!/login/login GET endpoint from a separate machine. I'm not sure if that's the expected behavior or not.

terrypacker, 2016-11-09:
CORS can be tested via the example page in the Dashboards Module.  You will need 2 running Mango servers.

Mango A: Hosting the page from the dashboards module at:
`<ma_home>/web/modules/dashboards/web/public/corsLogin.htm`

You will need to edit this file to add the ip address of Mango B.

If you have default settings it should be at this URL:

`http://localhost:8080/public-dashboards/corsLogin.htm`

Mango B: CORS Enabled with the hostname of Mango A in the allowedOrigins setting.

Then you should be able to use this page to login to Mango B from A.

terrypacker, 2016-11-10:
CORS is now working as expected.  The last outstanding problem is what to do about the override-web.xml file that anyone using CORS will have.  This file must be edited to remove the filter definition as it doesn't exist nor should it be used anymore.  There is a line in the RELEASE-NOTES about this.

terrypacker, 2017-03-28:
Closing as done.

closed 2017-03-28 by terrypacker