MangoAutomation/ma-core-public #677: New Data Point Details Page - 5000 Point value limit handling

Puckfist, 2016-03-11:
There is a hard limit on the new data point details page that restricts charting to less than 5000 points, and advises download if you exceed that. The chart is very nice to have!

After some discussion, it seems the intended workflow would have been to use the rollups to reduce the charted data, but this is nonobvious. There is no help bubble on the page, and the error message doesn't mention rollups.

The idea of downsampling arose, but downsampling would greatly reduce the integrity of the chart as being seeming random (based on position in the series) where rollups deliver information about what is happening in the data to reduce the total sample quantity.

This could be enhanced by offering more information to the user, notifying in the event of downsampling, or perhaps remembering the particular rollup for a point presuming the rollup is the typical way the user wishes to consume the data on the page.

resotek, 2016-03-15:
I've been thinking about this issue since last year. The preferred handling I came up with is a rollup that reduces the number of points but prints two or three datasets instead of one: either min and max, or min, average, and max. This would ensure data integrity. Since the statistics function calculates all these anyways, the computational load will not increase. The three numbers for each time slice could be represented by the "[Fill between two lines](https://www.amcharts.com/demos/fill-between-two-lines/ )" amCharts.

resotek, 2016-03-25:
Adding pan and zoom to the chart may go a long way to solving this issue.  According to [Line Chart with Scroll and Zoom](https://www.amcharts.com/demos/line-chart-with-scroll-and-zoom/#theme-patterns): 

> This is one of the most important features of charts, especially when dealing with large data sets.

See #689.

terrypacker, 2017-12-01:
We have added a Simplify option to the new Point Value endpoints in 3.3.0, this allows to request the raw data be decimated to roughly the desired size output.

See here for more information: [https://help.infiniteautomation.com/point-value-v2-endpoints/](https://help.infiniteautomation.com/point-value-v2-endpoints/)

closed 2017-12-22 by Puckfist