MangoAutomation/ma-core-public #678: Excel Report Swagger - /v1/excel-reports GET working?

Puckfist, 2016-03-13:
I attempted to "Try it now" after creating an running a report. i got a 500 error, a message about my query not being able to be null (but i see no fields in swagger to supply a query to this endpoint), and the attached stack trace.

[excel-endpoint-stacktrace.txt](https://github.com/infiniteautomation/ma-core-public/files/171049/excel-endpoint-stacktrace.txt)

terrypacker, 2016-03-13:
This is a known issue with the Swagger endpoints for RQL queries.  Since Spring doesn't seem to be able to parse RQL in the URL into a String we have to build the String ourselves from the HttpServletRequest.  This means that there isn't any annotations on the method that Swagger can use to build its UI.  This can probably be fixed but hasn't been important enough to bother with yet.

Since they are GET requests I usually just point the browser that the URL and append the query I want to test.

Puckfist, 2016-10-18:
The swagger endpoint or the RQL or something has alleviated this. Closing as fixed.

closed 2016-10-18 by Puckfist