MangoAutomation/ma-core-public #681: Core - Copy datasource fails when excessive data points

Puckfist, 2016-03-14:
Attached is a virtual data source with 5000 points. if one imports this data source, then attempts to copy it via the copy button, and tries to save the copy, this will park for a while. Seemingly nothing will happen. If you refresh the page, SQL errors will fill the section between the navigation bar and the data source table, and similar errors will be alerted to you if you attempt to export your configuration (or do almost anything).

Upon Mango restart, i found the copied datasource existed but had no points.

[output.txt](https://github.com/infiniteautomation/ma-core-public/files/172972/output.txt)

terrypacker, 2019-07-23:
This is likely due to the no-callback option for the Dao query to get the points during the transaction for the data source copy.

terrypacker, 2019-07-24:
I've confirmed that this is not an issue for the REST api

closed 2019-07-24 by terrypacker