MangoAutomation/ma-core-public #684: Core - Unit conversions, JFree charts and limit detectors

Puckfist, 2016-03-16:
If a value is converted to a different unit for display, it will be rendered on the JFree charts in the target unit. If limit event detectors are configured, it will depend on the unconverted value (the value which is actually saved in the database) for displaying the limit line on those same charts. This means the series can appear to be outside its 'normal range' (as defined by the limit detectors) while not being in a truly alarming state.

resotek, 2016-03-16:
The "Discard extreme values" feature on the point details tab is ambiguous: it is not clear to the user whether the "Discard low limit" and "Discard high limit" refer to the source or target units. This should be documented in the "Logging properties" help, or the relevant unit could be displayed to the right of the numeric field.

terrypacker, 2016-06-02:
Not changing for JFree charts, but will add notes to help or indication on page as to data point settings inputs and what units they will depend on.

Puckfist, 2017-10-02:
Around 227 of ImageChartServlet one can do something like this:

UnitConverter uc;
                if(markerPoint.getUnit() != null && markerPoint.getRenderedUnit() != null)
                    uc = markerPoint.getUnit().getConverterTo(markerPoint.getRenderedUnit());
                else
                    uc = Unit.ONE.getConverterTo(Unit.ONE);
                for (AbstractPointEventDetectorVO<?> ped : markerPoint.getEventDetectors()) {
                    if (ped.getDefinition().getEventDetectorTypeName().equals(AnalogLowLimitEventDetectorDefinition.TYPE_NAME))
                        ptsc.addRangeMarker(new ValueMarker(uc.convert(((AnalogLowLimitDetectorVO)ped).getLimit()), lowLimitPaint, limitStroke));
                    else if (ped.getDefinition().getEventDetectorTypeName().equals(AnalogHighLimitEventDetectorDefinition.TYPE_NAME))
                        ptsc.addRangeMarker(new ValueMarker(uc.convert(((AnalogHighLimitDetectorVO)ped).getLimit()), highLimitPaint, limitStroke));
                }

or we can close the issue....

closed 2018-01-23 by Puckfist