MangoAutomation/ma-core-public #686: DGLux Login Page - Bad Session Cookie

Puckfist, 2016-03-21:
If a user goes through the DGLux login page to access Mango, their session cookie will not be properly set, and all REST functions will respond with denied access to their pages (user, excel, logging, etc)

terrypacker, 2016-05-26:
This was fixed by ensuring that all DefaultPagesDef module definitions make it into the Spring Security mappings.  Changes were made by removing code from the web.xml config file and configuring the spring security filter programmatically at startup.  When testing be sure to check that any custom Dashboard modules with custom login pages still work.

Puckfist, 2016-05-27:
I do not believe it to be fixed, as I can navigate to /dglux/login.htm then to /users.shtm and the users page will not work.

Puckfist, 2016-05-27:
Currently it probably works if you set your default login page to dglux, but fails if you default page is still Mango but you navigate to the dglux login page either of your own willpower or under the direction of your administrator, then navigate back to Mango. There are two (three?) possible options: 1) DGLux's login page could be hardcoded directly into the Core, as it is the only module with a special login controller we are aware of and subsequent modules will be strongly encouraged to use the API, or 2) add a method to the DefaultPagesDefinition to specify valid login locations, or 3) assume nobody will navigate to the DGLux page and then back into Mango's UI. This option becomes increasingly bad as more Mango UI pages rely on the API

terrypacker, 2016-06-02:
Leave code so that this page MUST be the default login page.

closed 2016-06-03 by terrypacker