MangoAutomation/ma-core-public #687: Reports - Disk space leak

Puckfist, 2016-03-23:
There exist ways to orphan .data files in the reports NoSQL datastore.

It appears the issue is that the deleteReportInstance method deletes from the SQL database before the NoSQL database, so it has a difficult time getting a good points list back to delete. This affects deleting run reports from the reports page and reports configured for both schedule and email (which are auto-deleting).

To fix, move the ejt.update call in ReportDao::deleteReportInstance to after the logic for deleting from NoSQL.

terrypacker, 2016-04-13:
What about the method deleteReport? I guess the idea is that report instances can exist past the time that a report is deleted?

Puckfist, 2016-04-15:
Confirmed fixed, confirmed that idea you asked about is also correct, instances can outlive their template.

closed 2016-04-15 by Puckfist