MangoAutomation/ma-core-public #688: chart

resotek, 2016-03-25:


resotek, 2016-03-25:
Often datapoint values fall within a range, but go to zero when the equipment is disconnected or configured for maintenance. This causes extreme scaling of the data because the vertical axis is scaled using min/max over the time period.

Implementing [Zoomable Value Axis](https://www.amcharts.com/demos/zoomable-value-axis/) can solve this problem, as it enables the user to ignore extreme, outliar values, by zooming only to the data value range of interest. This could be implemented on at least the point_details.shtm page.

resotek, 2016-04-26:
Zooming the vertical axis would also be useful on the watchlist page graph. The image below is a good use case:
[image]

We are interested in the value during normal conditions, but the extreme negative value from when the equipment was taken offline for maintenance scales the entire graph so that the ripple voltage is almost invisible.

terrypacker, 2016-05-26:
Isn't this already implemented?  I can click and drag across the chart to zoom into a range.

Puckfist, 2016-05-27:
He was requesting zoom on the vertical Axis, which isn't possible in any of our charting.

terrypacker, 2016-05-27:
We need the latest 3.18 version of AmCharts to support this.

terrypacker, 2018-03-15:
Closing and opened infiniteautomation/ma-dashboards#138

closed 2018-03-15 by terrypacker