MangoAutomation/ma-core-public #689: Add a scrollbar to the horizontal (e.g. time) axis in amCharts

resotek, 2016-03-25:
According to [Line Chart with Scroll and Zoom](https://www.amcharts.com/demos/line-chart-with-scroll-and-zoom/#theme-patterns):

>All our serial charts (and actually XY chart too) can be zoomed-in, scrolled and panned. All you have to do is add a scrollbar to the chart. This is one of the most important features of charts, especially when dealing with large data sets.

This may mitigate or solve #677.

Puckfist, 2018-02-07:
While we may make the amCharts on the Data Point Details page in the new UI use the horizontal scollbar, we will not be modifying the old "New" data point details page. For ma-serial-charts used in new dashboards, the options parameter can pass to the chart that this scrollbar should be enabled. Like the following,

options="{chartScrollbar: {}}"

One could even override the data point details page to have this set, if one desired. I may soon cose the issue, but I will forward it to decision makers to see if they believe there should be a horizontal scroll if you zoom in on the data point detail page (and other UI spots) first.

jazdw, 2018-02-07:
We could add a setting to the UI settings page to enable the scroll bar thing if we want to put any effort into this.

Puckfist, 2018-02-08:
Was not deemed to happen on our details page

closed 2018-02-08 by Puckfist