MangoAutomation/ma-core-public #691: Ability to add arbitrary tags to data sources and data points

jazdw, 2016-04-06:
This would be handy for building dashboards so that you could query based on them. We are currently limited to searching by name, device name and datasource name. The point hierarchy could be used for some scenarios but can be too rigid.

Tags which I could see myself using would be, for example
- Site name
- Location

Existing properties such as "deviceName" could possibly be converted to tags.

terrypacker, 2016-05-25:
I don't think is a priority for 2.8.x.  I would like to put more thought into it before we make a decision to just add new members to the VOs.  This also aligns with the future concept of tags in Mango.  If you disagree please let me know.

jazdw, 2016-05-26:
Yeah push it back

Puckfist, 2018-01-03:
I tied this in so that scripting context points can now call p.getDataPointWrapper().getTags() but I didn't tie in any ability to set from a script

terrypacker, 2018-01-05:
TODO after release is allow tag queries from javascript like the REST api.

closed 2018-01-18 by terrypacker