MangoAutomation/ma-core-public #692: Apache POI - Upgrade Library

terrypacker, 2016-04-06:
If the temp directory or the folder 'poifiles' within it is deleted while Mango is running no Excel reports or Excel files can be created, a file not found exception will be thrown.  Upgraded to the latest dev version will fix this.

https://bz.apache.org/bugzilla/show_bug.cgi?id=57200

terrypacker, 2016-05-25:
Until this bug is resolved it would ok to use a different java.io.tmpdir as passed into Mango.

Puckfist, 2016-11-11:
The bug tracker referenced has someone commenting on 11-1-16 that they have resolved this.

terrypacker, 2016-11-11:
Moving to 2.8.x so we can release a new module in the near future.

terrypacker, 2017-01-24:
Closing as fixed, upgraded to 3.14

Puckfist, 2018-04-06:
There is now a 3.17

Puckfist, 2018-04-09:
Got this stack trace on trying to run a report for an unrelated reason,

ERROR 2018-04-09T12:38:34,821 [low-pool-4-thread-1] - getFileMagic() only operates on streams which support mark(int) 
java.io.IOException: getFileMagic() only operates on streams which support mark(int)
	at org.apache.poi.poifs.filesystem.FileMagic.valueOf(FileMagic.java:131) ~[poi-3.17.jar:3.17]
	at com.infiniteautomation.mango.excelreports.rt.ExcelReportWorkItem.execute(ExcelReportWorkItem.java:227) ~[?:?]
	at com.serotonin.m2m2.rt.maint.BackgroundProcessingImpl$WorkItemRunnable.run(BackgroundProcessingImpl.java:674) ~[classes/:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source) ~[?:1.8.0_111]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source) ~[?:1.8.0_111]
	at java.lang.Thread.run(Unknown Source) [?:1.8.0_111]

closed 2018-04-16 by Puckfist