MangoAutomation/ma-core-public #693: Add Name to MeasurelogicDTSCellConnectionServer Thread

terrypacker, 2016-04-06:
This will allow for easier detection and debugging.

terrypacker, 2016-04-13:
Consider using a thread pool to spawn connection handlers in the protocol connection server.  Code written to name the threads but a new library is not in the data source yet.

Puckfist, 2016-10-18:
I didn't see your naming of the threads in the 2.7.x branch. I have added this. I see how a connection pool could eliminate some Threads blinking to and fro in existence, but has this been an issue for a user?

terrypacker, 2016-10-19:
It wasn't an issue for a user, its just generally I good idea to put a bound on the number of threads that can be generated by a server socket listener.

Puckfist, 2016-10-25:
I have added a setting to restrict the client threads.

Puckfist, 2016-11-04:
Bounced forward if testing is required for the thread limiting.

closed 2017-01-03 by Puckfist