MangoAutomation/ma-core-public #697: Add a filter for points with no permissions on the user's data point permissions page

terrypacker, 2016-04-14:


terrypacker, 2016-04-15:
New checkboxes to enable and disable each column have been added as well as the option to use AND or OR type queries for each filter.

resotek, 2016-04-22:
Can NOT be used?

terrypacker, 2016-04-22:
@resotek NOT isn't an option.  The filter is used in a query on the data points table like this:
`
LOWER(dp.xid) LIKE ?
`

Where the input from the filter is used in place of the ?.

If you select And for a row then the query becomes:

`
LOWER(dp.xid) LIKE ? AND LOWER(dp.xid) LIKE ?
`

I was under the impression you were looking for a way to search for  points without permissions which can be achieved like this:

`
LOWER(dp.setPermission) LIKE null OR LOWER(dp.setPermission) LIKE ''
`

It would be possible to add NOT as one of the drop down options so that the options would be:

`
Disabled
And
Or
Not And
Not Or
`

Where Not And would generate a query like this:

`
LOWER(dp.setPermission) NOT LIKE ? AND LOWER(dp.setPermission) NOT LIKE ?
`

I could see that being useful to find points without certain permissions.

resotek, 2016-04-25:
@terrypacker Yes, that would be quite useful. I frequently receive requests for access by certain users who are receiving "unauthorized" messages. They rarely specify which points aren't working; they just want me to grant them more access. The first thing an admin would do is look for which points they do not have access to. A NOT filter should solve that problem.

Puckfist, 2016-05-27:
Neat! There are some issues, though. ^ is a permissible string in the permissions, if we care, so it is not possible now to specify a filter to match ^asdf as a permissions group.

Also, I got the stack trace attached a couple of times while applying / clearing permissions on 
[bulkPermissionsNegationsStackTrace.txt](https://github.com/infiniteautomation/ma-core-public/files/287255/bulkPermissionsNegationsStackTrace.txt)

terrypacker, 2016-05-27:
Any chance you have the exact query you were using at the time of the Exception?  That means our count query and the update query are different.  Also the type of database you were using would help as each has a slightly different query.

Puckfist, 2016-11-07:
While I was able to filter for blank permissions, "" using ^.+ , I was unable to filter for null permissions, instead getting the attached stack trace. I'm not sure how to create a null permission for a point, but it maybe had to do with #812 or perhaps being created by a data source tool.
[null-filter-stack-trace.txt](https://github.com/infiniteautomation/ma-core-public/files/575908/null-filter-stack-trace.txt)

Puckfist, 2016-11-08:
I get the same results filtering null and ^null - both display the points that have null as the value of the the permission.

Puckfist, 2016-11-08:
On the same not, null seems to break the "Not And" and "Not Or" logic...

terrypacker, 2016-11-08:
Fixed null and ^null to work.  

We need to test null queries for char columns specifically and other types of columns for this issue.

Added note to help about double ^^ when using NOT AND or NOT OR as well as specifically mentioned the '^' not character.

Puckfist, 2016-11-08:
The filters appear to be working nicely, and I see the help is thorough. I am getting the attached stack trace, though. To reproduce, navigate to the bulk permissions page with some points in your system, toggle the data point xid column off, then toggle it back on. Now, try to apply some permissions.

[bulk-permissions-apply-stack-trace.txt](https://github.com/infiniteautomation/ma-core-public/files/579629/bulk-permissions-apply-stack-trace.txt)

terrypacker, 2016-11-09:
Yes there were some issues surrounding the re-structure of the SQLStatement class, this is fixed and the only place it was currently used outside of the generic REST calls (which work fine) was here.

closed 2016-11-09 by Puckfist