MangoAutomation/ma-core-public #699: Upgrade Data Sources for 2.8.x

terrypacker, 2016-04-16:
For Polling Data Sources

This will allow cron patterns and quantization for all polling data sources.  The procedure is as follows:

VO Upgrade
1. Splice in PollingDataSourceVO by extending PollingDataSourceVO
2. remove private members pollPeriods and pollPeriodType and their getters
3. fix validate() method by removing 2 properties
4. change addPropertiesImpl to addSubclassPropertiesImpl (remove 2 properties)
5. change addPropertyChangesImpl to addSubclassPropertyChangesImpl (remove 2 properties)
6. fix readObject and writeObject to not use 2 properties
7. Fix JSONRead and JSONWrite to not use 2 properties

Model Upgrade
1. Extend PollingDataSourceModel
2. Remove pollPeriod getters

RT Upgrade
1. Use the following in constructor
`        if(StringUtils.isEmpty(vo.getCronPattern()))
        	setPollingPeriod(vo.getUpdatePeriodType(), vo.getUpdatePeriods(), false);
        else
        	setCronPattern(vo.getCronPattern());`

The following is the list of datasources to convert:
DNP3
ASCII
BrewersDB
TankDS
Data FIle
eGauge
HeartTransverter
PakBus
PID
SErial
SSH
TCPIP
EnvCan
Galil 
HTP
Internal
JMX
MBus
Modbus
OneWire
OPC DA
OpenV
Pachube
POP3
Scripting
SNMP
SQL
Virtual
BACnet

Joelhaggar, 2016-04-26:
These seem like the priority to me:
Modbus
BACnet
SQL
Virtual

These are less important but would be nice at some point in the future:
POP3
SNMP
SSH
TCPIP
HTTP
Serial

terrypacker, 2016-05-25:
Moving to Bugs as they can be implemented after the release.  Most of the high priority sources have been converted.

terrypacker, 2017-10-10:
I think we should do this before we implement the new Angular UI page for data sources.

Puckfist, 2018-02-07:
Not that it solves the issue necessarily, but you can currently run a data source on a cron by using the RuntimeManager.refreshDataSource(xid) method in the script. So, you can use a script as a cron or event wrapper for forcing data source polls. The downside being the polling rate of the data source still exists.

You can disable/enable to cause a data source to be quantized.

closed 2019-04-09 by terrypacker