MangoAutomation/ma-core-public #705: Make Event Detectors Modular

terrypacker, 2016-04-20:
Event detectors have been modularized from the backend but the edit page still only supports creating the 3 pre-existing types.

terrypacker, 2016-04-20:
Test DB Upgrades for all database types.
Test Create/Edit/Delete all detector types.
Test Emport Detectors.

Puckfist, 2016-05-31:
This definitely still has issues. I am going to rename this issue to just refer to ensuring old functionality, as per our earlier discussion there remains work to make modules' event handlers or detectors appear as usable.

Puckfist, 2016-05-31:
Operating on a Binary Point, I cannot add two event detectors at the same time. If they are the same type, I get a SQL error like the attached. If they are different types, I get the stacktrace attached.

In both cases, it is interesting to note that the XID is filled in for the first new event detector, and adding subsequent detectors will change the XID of the first new detector. This also manifests as the second detector's delete button deleting the first detector.
[eventDetector_diffType_castException.txt](https://github.com/infiniteautomation/ma-core-public/files/292139/eventDetector_diffType_castException.txt)
[eventDetector_sameType_sqlError.txt](https://github.com/infiniteautomation/ma-core-public/files/292140/eventDetector_sameType_sqlError.txt)

Puckfist, 2016-05-31:
The validation messages also go to the wrong div if there are multiple new event handles defined.

Puckfist, 2016-05-31:
If a Low or High limit detector is place on an analog point, the value for "use reset limit" and the "reset limit" itself do not survive opening the edit page again.

Puckfist, 2016-05-31:
As goes to the reset limit, you can set this via the importer and it will appear in the UI, so my suspicion is the problem is in marshalling the data from the UI and saving it.

Puckfist, 2016-06-01:
Databaes Upgrade12 failed with the attached trace.
[upgrade_stackTrace.txt](https://github.com/infiniteautomation/ma-core-public/files/294524/upgrade_stackTrace.txt)

Puckfist, 2016-11-02:
Relevant commit: https://github.com/infiniteautomation/ma-core-public/commit/11bde6cd6cc6c0266f13aa2515e4b2e6f17e9ca3

Puckfist, 2016-11-03:
I believe these issues have all been resolved.

closed 2016-11-04 by Puckfist