MangoAutomation/ma-core-public #708: Audit Tracking - Required Tracking not implemented

terrypacker, 2016-04-21:
Various members are not currently audited and to same time they are listed here to return to and implement:

DataSourceVO.alarmLevels

terrypacker, 2016-05-25:
This should be fixed with the new audit system.

Puckfist, 2016-06-03:
Got this error upgrading a Linux system:

ERROR 2016-06-03 17:37:59,346 (com.serotonin.m2m2.db.upgrade.Upgrade12.backupAuditEvents:283) - /opt/m280/backup/auditEventBackup.json (No such file or directory)
java.io.FileNotFoundException: /opt/m280/backup/auditEventBackup.json (No such file or directory)
        at java.io.FileOutputStream.open(Native Method)
        at java.io.FileOutputStream.<init>(FileOutputStream.java:213)
        at java.io.FileOutputStream.<init>(FileOutputStream.java:162)
        at java.io.FileWriter.<init>(FileWriter.java:90)
        at com.serotonin.m2m2.db.upgrade.Upgrade12$EventRowCallbackHandler.open(Upgrade12.java:329)
        at com.serotonin.m2m2.db.upgrade.Upgrade12.backupAuditEvents(Upgrade12.java:279)
        at com.serotonin.m2m2.db.upgrade.Upgrade12.upgrade(Upgrade12.java:173)
        at com.serotonin.m2m2.db.upgrade.DBUpgrade.checkUpgrade(DBUpgrade.java:98)
        at com.serotonin.m2m2.db.upgrade.DBUpgrade.checkUpgrade(DBUpgrade.java:34)
        at com.serotonin.m2m2.db.DatabaseProxy.initialize(DatabaseProxy.java:180)
        at com.serotonin.m2m2.Lifecycle.databaseInitialize(Lifecycle.java:633)
        at com.serotonin.m2m2.Lifecycle.initialize(Lifecycle.java:285)
        at com.serotonin.m2m2.Main.main(Main.java:116)

Dunno if more is achieved asking the File to createNewFile as oppposed to just opening a FileWriter and letting it attempt creation...

Puckfist, 2016-11-04:
The two listed in the original post are implemented and working, but there may be other items in Mango that are not audit'ed, like the point hierarchy. I am moving this on past 2.7.x for now.

terrypacker, 2018-04-04:
Closing this issue as this has been worked on in a separate issue that was closed.

closed 2018-04-04 by terrypacker