MangoAutomation/ma-core-public #709: Event Detectors send last 10 values but not necessarily the value that triggered the event.

Joelhaggar, 2016-04-22:
This could be due to pulling the recent values from the database rather than the cache.

terrypacker, 2016-05-09:
This is because when the event handler sends the latest 10 values it does request them from the cache, but the default cache size is 1.  This means that a request of the latest 10 values requires access to the database to load the latest 10 values into the cache.  At this point the cache will remain at size 10.

Basically the fix would be to set the cache size for that point to 10 if you need realtime values and are concerned about the delay from data being written into the database.

terrypacker, 2017-01-24:
Can we close this one now?

Puckfist, 2017-03-21:
Either that or merge the cache when expanding it

Puckfist, 2017-03-21:
Oh, or we can do the merge in the email handler rt, so that we don't affect the caching.

Puckfist, 2017-03-21:
Or we can expand the cache when the email handler RT is created.

terrypacker, 2017-05-23:
In PointValueCache when expanding the cache merge the existing cache with the retrieved values.

Puckfist, 2017-10-26:
It's a fair bit more work for the else case, but I think this would be similar to what we'd be looking for - if we decide to do this.

List<PointValueTime> loggedData = dao.getLatestPointValues(dataPointId, size);
                //older to the end
                List<PointValueTime> c = new ArrayList<>();
                Iterator<PointValueTime> cacheIterator = cache == null ? null : cache.iterator();
                Iterator<PointValueTime> loggedDataIterator = loggedData.iterator();
                PointValueTime nextC = cacheIterator != null && cacheIterator.hasNext() ? cacheIterator.next() : null;
                PointValueTime nextL = loggedDataIterator.hasNext() ? loggedDataIterator.next() : null;;
                while(c.size() < size) {
                    if(nextC == null && nextL == null)
                        break;
                    if(nextC == null) {
                        while(loggedDataIterator.hasNext() && c.size() < size) {
                            c.add(loggedDataIterator.next());
                        }
                        cache = c;
                        return;
                    }
                    if(nextL == null) {
                        while(cacheIterator.hasNext() && c.size() < size) {
                            c.add(cacheIterator.next());
                        }
                        cache = c;
                        return;
                    }
                    if(nextC.getTime() == nextL.getTime()) {
                      //only one value per time, prefer logged values
                        c.add(nextL);
                        if(cacheIterator.hasNext())
                            nextC = cacheIterator.next();
                        else
                            nextC = null;
                        
                        if(loggedDataIterator.hasNext())
                            nextL = loggedDataIterator.next();
                        else
                            nextL = null;

                    } else if(nextC.getTime() < nextL.getTime()) {
                        c.add(nextL);
                        if(loggedDataIterator.hasNext())
                            nextL = loggedDataIterator.next();
                        else
                            nextL = null;
                    } else if(nextC.getTime() > nextL.getTime()) {
                        c.add(nextC);
                        if(cacheIterator.hasNext())
                            nextC = cacheIterator.next();
                        else
                            nextC = null;
                    }
                }
                cache = c;

closed 2017-12-18 by Puckfist