MangoAutomation/ma-core-public #710: PointLinkVO needs proper validation

terrypacker, 2016-04-23:
In order to property validate xids the PointLinkDao needs to be converted and PointLinkVO needs to be updated to allow getDao()

terrypacker, 2016-06-03:
Please Test ReportDao/VO for xid validation.

Remaining:
PointLinkDao/VO - Delay until after 2.8.0 release
GlobalScriptDao/VO - Delay until after 2.8.0 release
ScheduledEventDao/VO - Delay until after 2.8.0 release
MaintenanceEventDao/VO - Delay until after 2.8.0 release

Puckfist, 2016-06-03:
I was able to save a report using a Point Link's xid, a data source's, and a data point's but not another report's.

terrypacker, 2016-06-03:
XIDs are only unique to the table of that type of item, so your test has passed.

Puckfist, 2016-06-07:
Not closed because those four are marked for delay?

I see they are validated in their Dwr save functions. Doesn't feel like quite the right place (I just noticed this in the other comment on Global Scripts validation). Unmarking as ready since this won't handle API cases, but they do check it at least.

terrypacker, 2017-12-04:
Moving to future features as this will need to be addressed when we expose these VOs via the REST api