MangoAutomation/ma-core-public #711: Data Source Startup - Better Handling of failed data sources

terrypacker, 2016-04-25:
It is possible for a data source to fail to start at startup.  This needs to be handled in one of the following ways:

A. Mango doesn't startup and terminates.
B. Logs the failure, raises an event and ensures that the enabled boolean is set to false.
C. Other?

terrypacker, 2016-06-02:
Raise Event, log it but don't set enabled to false.

Puckfist, 2016-11-17:
I vote for B, otherwise is it more difficult to see which data source it is. We'd have to consider the reasons for data sources failing to start, though. For instance, a missing module will fail and we wouldn't be able to fetch the results for the data source table to even bother displaying it as false, unless that's part of #135

Puckfist, 2018-03-19:
Missing lob in starting data point

Puckfist, 2018-04-23:
No events raised? Seems like the place to do that is inside the DataSourceGroupInitializer, specifically the DataSourceSubGroupInitializer

I think this issue probably originally pertained to the issue that if a data source failed to start in a given group's thread the thread would die and the other perfectly fine data sources don't start.