MangoAutomation/ma-core-public #713: Graphical Views, HTML Component Issue

Puckfist, 2016-04-28:
An error, "angular is not defined" is printed to the developer console after performing the following steps:

1) New graphical view, using version 1.6.5
2) Add HTML component, give it some content: <b>Hello HTML Component!</b>
3) Save the Graphical View
4) Refresh the page. Notice our text isn't there (neither is our component), and the add component select is also not functioning.

terrypacker, 2016-04-28:
We've added angular support to the Graphical View module recently, from what I can tell this means you must have a version of the Dashboards module that has mango-3.0 javascript as well as the Core's loaderConfig that has the mango-3.0 mapping.

jazdw, 2016-05-03:
Fixed via infiniteautomation/ma-modules-public@42e663a0eafe13a17e314577e706f9b847794b32

We should probably note somewhere that the material design components wont work.

terrypacker, 2016-05-05:
Just re-tested and this didn't fix the problem.

terrypacker, 2016-05-05:
Code was out of date, fix is good.

closed 2016-05-05 by terrypacker