MangoAutomation/ma-core-public #714: Graphical View, Shameless Promotion

Puckfist, 2016-04-28:
See WirelessTempHumSensor.java:34

This component creates an alert which proclaims itself to be a shameless promotion. It should be reviewed, and possibly removed entirely.

terrypacker, 2016-06-02:
Remove this component.

Puckfist, 2016-06-03:
I get an XHR failure on both systems I've tested, resulting in nothing being in the components list when I open a new view for editing. I have cleared my cache

Puckfist, 2016-06-03:
That comment was invalid, it was caused by my incorrect build target on the dashboards module, so I had no dashboards module.

Puckfist, 2016-06-03:
We probably only wish to remove the alert. Removing the whole component resulted in Stream Corruption at the Graphical Views page, the edit view page and the json export of the views for an upgraded system using the view.

Puckfist, 2016-06-07:
Looking at it now, I wonder if we want this as a widget to advertise our ability to design custom components. In part because it wasn't made by us, in part because its impressiveness isn't so self-evident to me, and in part because it's using the deprecated (or deprecating, at least) method for creating components.

Puckfist, 2016-10-25:
I have removed the alert from the component. It caused an alter whose first words are 'syntax error' The commit is here: https://github.com/infiniteautomation/ma-modules-public/commit/afc316b0409edca87f658afc1459673be5c9da94

Calling it closed.

closed 2016-10-25 by Puckfist