MangoAutomation/ma-core-public #718: Point Value deletion after changing Data Point data type.

terrypacker, 2016-05-03:
Currently a specific method is used deletePointValuesWithMismatchedType(id, type) but it seems that this is un-necessary as whenever this happens we will need to remove all of the legacy values for this point.  Also the new Cassandra module won't support this method.  

This means we need to ensure that point values with the wrong type cannot be imported or inserted by the runtime manager...

Also a confirmation dialog should be added to the UI when a data point is saved and its data type is changed.

Puckfist, 2016-05-27:
This has issues. It failed when I tried to change a running numeric data point with some history into a multistate. The history persisted across restarts and through switching data types many times. With both being on no change detectors, it spat out data type mismatches. There was a section that it failed to delete with retry, then threw this stack trace, saying the delete failed all its retries
[changeDataTypeStackTrace.txt](https://github.com/infiniteautomation/ma-core-public/files/287210/changeDataTypeStackTrace.txt)
:

terrypacker, 2016-06-02:
1. Create Virtual Data Source polling at 500ms
2. Add numeric point, save but don't change the view let poll for a bit to generate some data
3. Change data type, save and see validation error on page for text renderer
4. See poll errors about mismatched data type

It seems that the new point locator is making it into the Virtual Data Source even though the validation of the Text Renderer is failing.

This is because the data point is saved in the server side state and it is also the exact same object that is loaded into the Virtual data source.  So the values are set into the polling point and then validated, but by that time the data source is already using that point!

closed 2016-11-04 by Puckfist