MangoAutomation/ma-core-public #722: NoSQL Shutdown Hook for Corruption Scanner

terrypacker, 2016-05-12:
Add code to terminate the corruption scanner when Mango is shutdown during a startup.

terrypacker, 2016-05-12:
This will require some thought, the scanner is only created and run during the constructor of the database.  So there is no way to terminate the database until this has finished. A few options are:

- Move scanner into initialize method
- Have Mango not perform a corruption scan in the constructor via the config scan=false then use an env.property to choose whether to perform the scan after the constructor is run.

Puckfist, 2017-10-03:
Points are corruption scanned live so if one is worried about this they can just disable corruption scan on startup, then there will be no single large corruption scan task, and the only tasks that could delay shutdown will be single points' fixes.

closed 2017-10-03 by Puckfist