MangoAutomation/ma-core-public #725: Meta Point Timestamps Sent into Javascript Engine Wrong

terrypacker, 2016-05-18:
The runtime of the script and the timestamp to use for the data point are swapped.  

The runtime of the script should be close to clock-time and the timestamp should be the time that the task was originally scheduled to run.

JavaScriptPointLocatorRT:50 flips arguments names runtime and timestamp in its call to the executor, but it seems the executor uses the correct one for the new value

terrypacker, 2016-05-25:
According to what I am seeing, this is all ok.

Puckfist, 2016-05-27:
Yeah, this was only ever about refactoring the names, I found it weird something called 'runtime' in one was being passed as 'timestamp' to the other function which has a variable 'runtime.' It's not so bad and does perform correctly, and I understand it's for history generation/delayed execution and the like. Closing.

closed 2016-05-27 by Puckfist