MangoAutomation/ma-core-public #732: TimeoutDetectorRT - Ensure Task IDs are correct

terrypacker, 2016-05-26:
Since XIDs are not unique, it is a compound key of sourceId vs Xid be sure not to use the XID.

Puckfist, 2016-11-17:
This can be closed? I placed a system.out.print at the getTaskId call, then setup a two data points on the same data source with two no change detectors with the same XID. My testing indicated that the tasks have different task IDs even for the same detector between runs. It's a new TimeoutDetectorRT for every run TimeoutDetectorRT:74

terrypacker, 2016-11-18:
This has to do with the 2.9.x branch's Task handling with the Serotonin Timer 3.0 code.  At this point we should leave it open as we haven't decided to dump that timing code just yet.  Removing high priority tag since its not in our current release stream.

terrypacker, 2017-03-28:
Closing this as the timing code isn't going to change for this anytime soon.

closed 2017-03-28 by terrypacker