MangoAutomation/ma-core-public #734: Meta Points updating on Context multiple times.

terrypacker, 2016-05-26:
This causes data to be generated multiple times at the same timestamp and is inefficient.

terrypacker, 2016-05-27:
This change has been implemented to only run the script on the first time the point is notified at a given timestamp and disregard all other notifications at that exact timestamp.

We need to discuss it and see if that is really the desired behavior.

terrypacker, 2016-06-02:
Currently there is a potential for a meta point to run on context and not all of its points to be updated to their current values in a data source poll.  Consider one virtual source with 3 points.  Point1 is update on context but the meta script uses all points.  When the first point is updated and the script is run, the other points still have the value from the previous poll.

terrypacker, 2016-06-02:
Maybe imply that an execution delay should be set to allow the data source poll to finish?

terrypacker, 2016-06-03:
We could also enforce a validation rule where if a point has context points from the same data source there must be some execution delay...

Puckfist, 2016-06-08:
I still feel there is the possibility to solve this in in having a density limit. Maybe something like....

PLRT gets update notification 1, executes script, produces prospective value 1, schedules save value task for X ms from now

PLRT gets update notificatoin 2, executes script, produces prospective value 2, task is already scheduled for save, does nothing
....
PLRT gets update notification y --> propect value y, task is scheduled already

Task fires, saves value y.

PLRT gets update notification y+1 --> prospect value y+1, no task so schedules saveValueTask for X ms from now
....

Thoughts?

Puckfist, 2016-11-04:
Removed volatile keyword as unnecessary: https://github.com/infiniteautomation/ma-modules-private/commit/16d83eab17173b22056d1874639941429712a293

closed 2016-11-04 by Puckfist