MangoAutomation/ma-core-public #738: Core - Polling data source model, object structure

Puckfist, 2016-05-27:
When exporting a polling data source on the emport page, you will have properties named "updatePeriods" and "updatePeriodType" while using the API (tested via swagger) you will receive a "polling" object with properties "periods" and "type"

In my opinion this needs to be consistent with the older version, the "updatePeriods" and "updatePeriodType"

terrypacker, 2016-05-27:
The models are different from the JSON Export since they are not used the same way.  Also if we change the models it will break the code for anyone who uses them in javascript.

Puckfist, 2016-05-27:
This makes me wonder how else one JSON representation of the object varies from the other. I feel strongly they should conform whenever possible.

closed 2017-07-26 by Puckfist