MangoAutomation/ma-core-public #740: None Level Events HTML5 Page

terrypacker, 2016-05-30:
It appears that the None level events are being sent out on the web socket but not when the EventsRestController.getActiveSummary endpoint is called.  It appears they are returning to normal and the message makes it back to the page and shows negative totals for these events.

terrypacker, 2016-05-31:
Code changed to not send notifications about none level events returning to normal.  Not sure if this is what we want but will revisit after a discussion of this issue and #602

terrypacker, 2016-05-31:
Something else to note here is that since we are not generating user events for NONE level events they are not returned via the events report page's queries.  The events report query joins events to the user event's table.

Puckfist, 2016-11-07:
It still bothers me a little that the sort order on the html5 pages is flipped from that of the DWR pages (meaning highest alarm level at the top of the set of active alarms, while the DWR pages puts the highest alarm level at the bottom of the header

Puckfist, 2016-11-07:
I have updated the ordering, and would be ready to close this. https://github.com/infiniteautomation/ma-core-public/commit/77e47eb34b52e1d997194e555f9432256a33f660

terrypacker, 2016-11-07:
Looks good to me.

closed 2016-11-07 by terrypacker