MangoAutomation/ma-core-public #748: Core Upgrade Confirm Modules Downloaded

terrypacker, 2016-06-02:
A core can be upgraded but not all modules if something goes wrong.  This leaves data sources un-upgraded and they won't load on the restart.

Puckfist, 2018-05-03:
Maybe we should try to load the module.signed file and see if we're able to do so?

terrypacker, 2018-05-04:
Sounds good, just make sure this only happens on the download (not startup) as we will want to retain the ability to start with unsigned modules.

Also make sure that this code is used in both the DWR and REST actions which are shared via the DWR code.

Puckfist, 2018-05-23:
https://github.com/infiniteautomation/ma-core-public/commit/4a4eeb5597bc1a9f49ba68ebbbad6a0a76174811
https://github.com/infiniteautomation/ma-core-private/commit/781265a6bdf3089ef012befb198aba19407e0071

Puckfist, 2018-05-23:
Seems like this would be best to do in a minor release

Puckfist, 2018-09-11:
I believe as of 3.5 this is happening with trying to check the version in the module.signed file?

terrypacker, 2018-09-11:
That is correct, this has been lightly tested but should be reviewed and then closed.

closed 2018-11-06 by terrypacker