MangoAutomation/ma-core-public #751: Ordered Task Queue Sizes

terrypacker, 2016-06-03:
Tasks are queued based on ID.  If this queue gets full then tasks are rejected.  In the legacy Timer system nothing was ever rejected for this reason as it was not possible.  We must decide on queue sizes for all task types or some way to adjust the queue sizes during runtime to avoid rejections (unless rejections are desired as is the case in something like a backup).

We could dynamically expand queue sizes if tasks are rejected, although this would require some tasks to get rejected.  Since we know the type of rejection we could keep increasing the size of the queue so long as the pool isn't rejecting tasks.

Note that point event notify work items are at high risk here since a data point can have many listeners ...

Puckfist, 2018-01-11:
I'll just drop this link here: https://help.infiniteautomation.com/internal-task-performance/

terrypacker, 2018-03-15:
Closing as reviewed.

closed 2018-03-15 by terrypacker