MangoAutomation/ma-core-public #757: NoSql - Windows MBB issues

Puckfist, 2016-06-07:
On a Windows machine, using the MappedByteBuffer for the NoSql fails on conversion or corruption scan. My test was: 

1. Create a brownian point
2. let it build a little history
3. end Mango
4. Open shard, change all 0xfe to 0xbb (corrupt everything)
5. Mark database dirty
6. Start Mango

This led to a ceaseless set of delete failed errors. I repeated the test on a Debian system, and it deleted the file plenty happily.

Windows is also failing to use the MappedByteBuffer in performing a conversion as part of the boot process, as it cannot delete files recently read by a MBB, it seems. Brief googling reveals one can either use the sun Cleaner to do an 'unmap' of sorts, or we could restrict stream use in those processes by Windows users.

terrypacker, 2016-10-28:
Code was added to ensure MappedByteBuffer streams cannot be used on windows.

terrypacker, 2016-10-28:
Add support for Windows by reviewing and testing the 'cleaner' method with in the DB Lib.

terrypacker, 2016-11-11:
Since performance on Windows is poor with the NoSQL database in reverse mode I am bumping the priority for this.

Puckfist, 2016-11-21:
I will likely be the one testing this, but I committed these changes because they appear to be working (tried purge, played with data import module). I would appreciate checking those few modifications I just made for sanity.

Puckfist, 2016-11-21:
Also I have heard tell this could be problematic for JDK 9: https://blogs.oracle.com/java/get-ready-jdk9

closed 2017-04-19 by Puckfist