MangoAutomation/ma-core-public #764: NoSQL Purge Improvements

terrypacker, 2016-06-13:
The feedback for the purge is inaccurate in that the count of values between dates is returned but the actual purge is done on a shard basis and would rarely delete exact the number counted.

In addtion, purge periods of less than 12 days will not be possible.  I suggest implementing a true delete for purge periods of less than 12 days.

Puckfist, 2017-03-06:
This has been added. Now all purges less than about 24.5 days will be accurate, purges longer than that can be accurate if the user has opted-in to purge within shards.

closed 2017-03-06 by Puckfist

Joelhaggar, 2017-03-30:
A potential conciseness of this is people that have set their purge period to 2 weeks with the expectation that they really always have about 5+ weeks of data.  I can think of at least one large customer that has their purge settings this way.  The less than 24.5 day purge will also put extra system load so for a large system this could cause an unexpected issue.