MangoAutomation/ma-core-public #768: Core - Database conversion - pointEventDetectors table

Puckfist, 2016-06-28:
It needs to be removed from the list of core table names in the conversion tool, DBConvert:74

Puckfist, 2016-06-28:
This problem probably also applies to the Graphical Views module, since the graphicalViewUsers table is not in the create tables but is added to the set of table names in the SchemaDefinition. This causes conversions to fail. We should remove it from GraphicalViewsSchemaDefinition:34

Puckfist, 2016-06-30:
The createTables-MySQL for the publishers table is missing the "tags" column that is present in the createTables-H2

Puckfist, 2016-06-30:
I have implemented the changes in these three comments.

terrypacker, 2016-11-03:
Removing the tags column from MySQL and Derby schemas as this is not a column we use.

terrypacker, 2016-11-03:
Closing and opening new issue about DBConvert missing tables.

closed 2016-11-03 by terrypacker