MangoAutomation/ma-core-public #774: Excel Reports don't handle missing data points well.

Joelhaggar, 2016-07-12:
If you create a report with data points and then delete those data points you won't be able to save your Excel Report again.  Also if you try and import JSON of the Excel Report with an error that doesn't tell you what the missing XID is:

Excel Report 'XLT_474461': 'JsonException writing property 'timeSeries' of class java.util.List', caused by: 'JsonException writing property 'points' of class java.util.List', caused by: 'Report point missing attribute 'pointXid''

Puckfist, 2016-10-18:
I'm not sure what the direction to take this is. Currently, a null pointer is thrown from PointRange:149 if you try to save through the UI. This could probably be fixed by returning on the vo == null check earlier in the function. Also, it seems PointRange has some dead code in it, as though the return; at 155 was intended to be inside the if at 153. This should probably be reviewed.

As to the JSON, it cannot report the XID of the point, but it does tell you the XID for the Excel Report this applies to. Perhaps it would be preferable to not export this range? I wouldn't think so, I would perhaps prefer the handling for the JSON error as it is.

terrypacker, 2016-10-28:
Changes applied from Puckfist's comment, ready to test.

closed 2016-10-31 by Puckfist