MangoAutomation/ma-core-public #783: Meta Data Source NPE

terrypacker, 2016-07-27:
It appears that if no values are returned from a script then you can get a NPE:

ERROR 2016-07-27 10:14:14,059 (com.serotonin.m2m2.meta.MetaPointLocatorRT.execute:300) -  
java.lang.NullPointerException
	at com.serotonin.m2m2.rt.script.ScriptUtils.coerce(ScriptUtils.java:185)
	at com.serotonin.m2m2.rt.script.ScriptExecutor.getResult(ScriptExecutor.java:143)
	at com.serotonin.m2m2.rt.script.CompiledScriptExecutor.execute(CompiledScriptExecutor.java:86)
	at com.serotonin.m2m2.meta.JavaScriptPointLocatorRT.executeImpl(JavaScriptPointLocatorRT.java:50)
	at com.serotonin.m2m2.meta.MetaPointLocatorRT.execute(MetaPointLocatorRT.java:291)
	at com.serotonin.m2m2.meta.MetaPointLocatorRT.contextPointUpdated(MetaPointLocatorRT.java:175)
	at com.serotonin.m2m2.meta.MetaPointListener.pointUpdated(MetaPointListener.java:51)
	at com.serotonin.m2m2.rt.dataImage.DataPointEventMulticaster.pointUpdated(DataPointEventMulticaster.java:93)
	at com.serotonin.m2m2.rt.dataImage.DataPointEventMulticaster.pointUpdated(DataPointEventMulticaster.java:93)
	at com.serotonin.m2m2.rt.dataImage.DataPointEventMulticaster.pointUpdated(DataPointEventMulticaster.java:93)
	at com.serotonin.m2m2.rt.dataImage.DataPointEventMulticaster.pointUpdated(DataPointEventMulticaster.java:93)
	at com.serotonin.m2m2.rt.dataImage.DataPointRT$EventNotifyWorkItem.execute(DataPointRT.java:572)
	at com.serotonin.m2m2.rt.maint.BackgroundProcessing$1.run(BackgroundProcessing.java:54)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

terrypacker, 2016-07-27:
I think this only happens when a point is used in the script that has never had a value in it.

Puckfist, 2016-11-14:
I think you fixed this? https://github.com/infiniteautomation/ma-core-public/commit/f8144b9f99cbb6abfa25efbdc446977da431d592

closed 2016-11-14 by Puckfist