MangoAutomation/ma-core-public #793: Get point values for multiple points from REST API

jazdw, 2016-08-05:
Will allow easier and faster querying of point values when charting point values for multiple points.

jazdw, 2016-08-08:
Partially implemented at 4582484c49dbce42b7c97055bc9de3f76a9d3e66
Doing the time range query is more complicated, @terrypacker  do you want to look at implementing this?

terrypacker, 2016-11-01:
This is done in issue #806

terrypacker, 2016-11-10:
We will have to look at performance on this:

Changes to the PointValueFacade:

1.  Add method getPointValuesBetween with Callback
2.  Add method getLatestPointValues with Callback
3.  Modify the getPointValuesBetween (initial & final) to use Wide Queries
4.  Add method getPointValuesBetween (initial & final) to use Wide Queries with Callback

We also will need to create a MultiPointValueFacade that uses new queries to be added to the PointValueDao:

1.  getLatestPointValues with Multiple IDs.

terrypacker, 2016-11-11:
Testing uncovered that if you make this request for points that are all within the same datasource and during a poll of said datasource it is possible to get values that don't line up.

For example:

If you request the latest 10 values and the datasource has 2 points it is possible that the array may contain 12 items and the first item would have only point A's value in it at a time 10 polls ago while the last item would have only point B's value at the currently polling poll.  This assumes that the points are polled in order of A,B within the datasource.

terrypacker, 2016-11-11:
Moving to 2.8.x to fix outstanding problems.

terrypacker, 2016-11-11:
This brings up the question of if we want to consider a different type of PointValueCache that is based on all the points within a data source and is updated with a lock and can only update all points at a time.

Puckfist, 2016-12-01:
I think we may consider either reworking these endpoints or adding some that support for POST, since the URL length restriction governs the restrictions on request size.

jazdw, 2017-04-03:
@Puckfist yep agree, I would suggest just using POST but it shouldn't hurt to just leave the old endpoint here.

Puckfist, 2018-01-23:
The post endpoint for large volumes of points was added to the V2 point values rest controller. So, closing

closed 2018-01-23 by Puckfist